openwch/ch32v003 #3: How is the flash programmed?

jonsmirl, 2022-10-26:
Does CH32V003 the on-chip boot code have the ability to flash the device, or is the only way to flash a new CH32V003 is via SWD?  We would like to attach this chip to a larger CPU and allow that larger CPU to update the flash in the CH32V003.

fuxiaoxiong, 2022-11-03:
Unfortunately  WCH_Link (SDI) and BOOT protocol is not open, WCH_Link (SDI) but not SWD is the only way to flash CH32V003.

jonsmirl, 2022-11-03:
Apparently Mounriver has released the protocol. https://github.com/kprasadvnsi/riscv-openocd-wch
But this is not in a form where it can be used by a host CPU. Addition work is needed to extract out the minimal programming library.

martinribelotta, 2022-11-03:
A reverse inginering of WCH (ch32v) SWD protocol is made by @fxsheep here https://github.com/fxsheep/openocd_wchlink-rv/wiki/WCH-RVSWD-protocol
The flash algorithm for varios wch parts was found in @kprasadvnsi repo on https://github.com/kprasadvnsi/riscv-openocd-wch/blob/master/src/jtag/drivers/wlink.c#L110-L337
Maybe a new openocd with ch32v003 support is required but the MounRiver guys are receptive if you send an e-mail to support@mounriver.com
But this require a lot of work for obtain a full floss toolchain to work...

@jonsmirl In my view, you have two ways to go:
 - Obtain a wlink and sniff the SWDIO during programming (hard way)
 - Implement a bit-bypass in your host cpu using one SWDIO input to the wlink cable and the slaves ch32v003 SWDIO pins

jonsmirl, 2022-11-03:
This is for a commercial product, so we don't have the luxury of spending a lot of time on open source code development. If there is no simple solution to the host being able to flash the chip we'll just continue using Padauk OTP parts which are not upgradable.

fuxiaoxiong, 2022-11-04:
@jonsmirl  So the production is 

**1. using a large host CPU is MCU to flash slave CH32V003, the host and slave MCU on the same PCB board. And you want to use host CPU loading programmer to CH32V003 by ETH , WIFI  etc. at real time .**
_**Ans: Write a private protocol ISP**_ 
**2. The host CPU or MCU characters a load programmer tools. CH32V003 is the only MCU on the PCB board.**  
_**Ans: Use WCH_Link, data transfer  just one-wire completed loading , thus three pad （VCC, GND, DATA）should be reserved on the PCB board**_

martinribelotta, 2022-11-16:
>  If there no PC to do that, WCH have a loading tools with out PC

That sound interesting... some plans to publish this with the regular HAL/Examples?

martinribelotta, 2022-11-16:
An other cuestio:

Is ok if I make a reverse engineering of the CH32V3xx and CH32V003xx SWD/SDI protocol?

josuah, 2022-11-28:
Here is another discussion with multiple engineers interested in an open-source tool-chain to program the chips:

https://www.eevblog.com/forum/microcontrollers/wch-$0-10-risc-v-mcu/?all

claudioolmedo, 2023-02-16:
Is it possible to use another CH32V3xx made with serial communication firmware to record another CH32V3xx?
If it's that cheap you can have two chips for low-cost denveloper boards.
Without having to buy the official WCH-LinkE, can anyone confirm if it's possible?

jonsmirl, 2023-02-16:
WCH Link clones are $1.20.   
https://www.aliexpress.us/item/3256804314210866.html

fxsheep, 2023-02-17:
Since RVSWD is a synchronous protocol, I think it can be implemented as a custom transport+ bitbang driver in OpenOCD and reuse existing bitbang SWD hardware. This possibly reduces cost to 0.

martinribelotta, 2023-02-17:
From Patrik twitter:
[image]

If you can wait, they will post the documentation of the 1-wire swd protocol

zoobab, 2023-02-21:
There is now a simple libusb program minichlink to flash it:

https://github.com/cnlohr/ch32v003fun

andelf, 2023-02-28:
Checkout https://github.com/ch32-rs/wlink 😝

Dreagonmon, 2023-02-28:
It seems the recent commit contains an example implement of using the one-wire procotol: https://github.com/openwch/ch32v003/tree/main/CH32V003_1Line_Base_on_CH32F103

perigoso, 2023-02-28:
> Checkout https://github.com/ch32-rs/wlink stuck_out_tongue_closed_eyes

ooh that might be handy, I'm planning to implement this debug protocol in Black Magic Debug, see blackmagic-debug/blackmagic/pull/1399, though for now I'm only working on the undocumented RVSWD (see also https://perigoso.github.io/rins and https://github.com/perigoso/sigrok-rvswd for RVSWD related things)

yrabbit, 2023-04-13:
> WCH Link clones are $1.20. https://www.aliexpress.us/item/3256804314210866.html

These are indeed cheap clones, but they are Link clones that cannot program the ch32v003 - you need LinkE for that

krasin, 2023-06-10:
FIY: https://github.com/aappleby/PicoRVD is an example of how to flash CH32V003 from an RP2040. The code is fully open-source and should theoretically be adaptable to other uses, such as a host MCU flashing a peripheral CH32V003.

qiyongshuang, 2023-12-14:
I noticed that there is an USART_IAP program in EVT（CH32V003EVT\EVT\EXAM\USART_IAP）.The program is placed in the BOOT area. We can use the USART port to write Flash.