openwch/ch32v003 #9: Example for I2C is not working

waiweng83, 2023-02-24:
The I2C example is not working. The Master always hold the SCL pin low.
I've added external 4K7 pull up resistor to both PC1 and PC2 but it's still the same.

qiyongshuang, 2023-02-27:
I want to know which EVT does not work or all I2C routines do not work.

waiweng83, 2023-02-27:
> I want to know which EVT does not work or all I2C routines do not work.

I tried "I2C_7bit_Mode" and it's not working. Didn't try on other examples.

qiyongshuang, 2023-02-27:
In "I2C_7bit_Mode“, first reset the slave board, then reset the host board.  "I2C_7bit_Mode“rountine sends and receives multiple arrays, and the result printing will have a delay.
[image]

closed 2023-12-14 by qiyongshuang