openwch/ch32v003 #11: NRST option byte defaults differ from Reference Manual.

MartinEssink, 2023-02-28:
The default value of the option bytes (See RM 16.5) seem to have the reset pin enabled by default, the value is 11010111 where bit 3 and 4 are RST_MODE (OB_RST_EN_DT1ms). In the reference manual it is stated that the default value is 11 (OB_RST_NoEN).

qiyongshuang, 2023-03-01:
Thanks for finding the problem. RST_MODE is 10 bit (12ms) by default. We will modify it later in the manual. If you want to use PD7 as a common pin, you can set it by using WCH_LinkUility.
[image]

MartinEssink, 2023-03-02:
Thank you for confirming this, and updating the reference manual!

closed 2023-03-02 by MartinEssink