openwch/ch32v003 #17: Documentation: CH32V003RM.PDF v1.4: Does not match with physical device settings

siuyin, 2023-05-23:
The reference manual: http://www.wch-ic.com/downloads/CH32V003RM_PDF.html  
Section 3.4 Register description (page 17 on v1.4) states  
R32_RCC_CFGR0 has a reset value of 0x00000000.

This is incorrect. On a physical device, the reset value was found to be 0x00000020.

Please update the documentation or issue an errata for the CH32V003 part.

qiyongshuang, 2023-05-26:
What method are you using to read the registers? I read it 0X00000000

siuyin, 2023-05-26:
The clock frequency immediately after reset should be 24MHz but the batch of CH32V003 chips I have run at 8MHz.

qiyongshuang, 2023-05-26:
Did you see the print or the MCO pin? Which CH32V003 chip do you use? I use ch32V003F4P6 do not have the problem .Do you use our EVT as test project?

siuyin, 2023-05-26:
I used CH32V003F4P6 and measured system clock frequency by switching an LED on and off with a logic analyser.
This is the test code I used:https://github.com/siuyin/ch32v003fun/blob/871a7350aac339c28b52cc4c08f3cd86e4bc2701/examples/minimal_blink/minimal_blink.c#L41

The setup code is: https://github.com/siuyin/ch32v003fun/blob/871a7350aac339c28b52cc4c08f3cd86e4bc2701/ch32v003fun/ch32v003fun.c#L766

SETUP_SYSTICK_HCLK is: https://github.com/siuyin/ch32v003fun/blob/871a7350aac339c28b52cc4c08f3cd86e4bc2701/ch32v003fun/ch32v003fun.h#L4856

qiyongshuang, 2023-05-26:
I grabbed the MCO pin through the oscilloscope, and the system frequency shows 24M. I suspect your test program have question.

siuyin, 2023-05-26:
R32_RCC_CFGR0 immediately after reset = 0x20. Causes HCLK=8MHz.
According to documentation CFGR0 should be /1 but is actually /3.

[image]

siuyin, 2023-09-27:
Indeed UART Enable should enable when set.

qiyongshuang, 2024-01-09:
Thank you very much, We will change it"uint32_t tmp = 0x00010000;"