openwch/ch32v003 #22: LCKK bit of GPIOx_LCKR defined incorrectly in ch32v00x.h

basilhussain, 2023-08-12:
In the EVT/EXAM/SRC/Peripheral/inc/ch32v00x.h header, the GPIO_LCKK definition is incorrect for CH32V003 hardware. It is defined as being in bit 16, but it is actually in bit 8. Also, it thus follows that LCK8 through LCK15 do not exist.

I propose that the definitions for GPIO_LCK8 through GPIO_LCK15 are removed, and GPIO_LCKK is modified as follows:

```c
#define GPIO_LCKK                               ((uint32_t)0x00000100) /* Lock key */
```

qiyongshuang, 2023-12-14:
Thanks for your suggestion.we will fix it .
[image]

qiyongshuang, 2024-01-15:
Please refer to the latest manual.

closed 2024-01-17 by basilhussain