openwch/ch32v003 #34: How to start with CH32V003  ?

xcom169, 2024-02-16:
Hello All,
I bought and soldered the CH32V003  product, however when I plug it onto a USB port nothing happens.  I can't see any now device in Windows.  On Linux there is nothing in dmesg.  Only a blue LED is blinking on the board.  How to start the device?  Or is it faulty?

xcom169, 2024-02-16:
I also tried Ardunio IDE but nothing happens with it either.

mrx23dot, 2024-02-16:
I don't think there is any USB chip on that board
https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Fae01.alicdn.com%2Fkf%2FS79da539b6b564dbc92dd1bc2b869e99bj%2FWeact-ch32v003f4u6-ch32v003f4-ch32v003-ch32v0-ch32-RISC-V.png&f=1&nofb=1&ipt=ed2f085d369bc700fec98f3dc68f861b0e5e3f9859740a8ccda75c844c0f8267&ipo=images
you need an external USB to serial converter.

xcom169, 2024-02-16:
Hello.
Thanks. Do you mean that USB-c is just for power nothing else?

mrx23dot, 2024-02-16:
CH32V003 doesn't have usb, you need another USB IC for that, if it's not on the board then USB is only used for power.

xcom169, 2024-02-17:
Thank you Árpád!

I have a USB to TTL-serial converter at home.
On the board are the DIO,NR, GND, VCC pins the serial pins?
Should I provide the 5V power over USB or via serial ?

mrx23dot, 2024-02-17:
You need a WCH linkE programmer if you want it do something, it also has built in UART.

https://www.tindie.com/products/adz1122/ch32v003-risc-v-mcu-development-board/
https://www.youtube.com/watch?v=YeclCYnlnp0

closed 2024-08-06 by qiyongshuang