openwch/ch32v003 #35: No firmware for Micropython for this ch32v003.

anandr16910, 2024-02-20:
Running and flashing this microcontroller board with MicroPython requires firmware associated with this part number **CH32V003**. Unfortunately, [https://micropython.org/download/] or [https://github.com/micropython/micropython?tab=readme-ov-file#supported-platforms--architectures] does not have firmware for this Microcontroller.

mrx23dot, 2024-02-20:
micropython on 2KB RAM? :D

anandr16910, 2024-02-20:
is it possible on 2kb ram

martinribelotta, 2024-02-20:
> is it possible on 2kb ram

From the repo readme:

> Microcontroller targets can be as small as 256kiB flash + 16kiB RAM, although devices with at least 512kiB flash + 128kiB RAM allow a much more full-featured experience.

Maybe a old basic interpreter (with tokenization) wold be posible... but not micropython in the actual implementation

closed 2024-10-18 by qiyongshuang