openwch/ch32v003 #41: Problem in interfacing bme280 sessor with vsdsquadron

SwapnanilChakraborty, 2024-05-05:
Since vsdsquadron not compatible with arduino or esp8266 libraries then what libraries we should use to send the data read by bme280 sensor to vsdsquadron  by using i2c communication between sensor and vsdsquadron board which uses the CH32V00X MCU

SwapnanilChakraborty, 2024-05-06:
Actually I need to interface bme280 sensor with vsdsquadron for reading the sensor data using i2c for that after looking at the data sheet of vsdsquadron I found the PC2 pin is for SCK and PD1 for SDA I made the connections between bme280 and vsdsquadron but when it comes to coding part since vsdsquadron is not compatible with arduino libraries I am not able to get the coding idea  and the vsdsquadron uses A CH32V00X MCU And noneos-SDK framework

qiyongshuang, 2024-05-09:
Do your want to communicate with vsdsquadron? if yes ,your should contact the manufacturer of the  vsdsquadron product.

closed 2024-06-17 by qiyongshuang